Syzygyx/DroneWidgets #4: Fuel gauge

BojanKverh, 2015-05-25:
Daniel,

I finished the fuel gauge widget, please check if everything is alright, when you have time.
I will now continue with the weight gauge.

Few questions:
1. How is the mission duration calculated (for the Distance/Duration gauge in Mission planning widget)
2. Is there a top-down vehicle view (slide 5 in the presentation) available somewhere or the application should create it from scratch?

Thanks for your help.

dcmcshan, 2015-05-25:
Cool.  I’ll take a look.

1. Max mission duration (for now) is just (Fuel Wh / Hover Power) as in spreadsheet:  https://docs.google.com/spreadsheets/d/18lJFzTITGgSfx1OGH_KkDWdKXgkGTt5et7Lv4X5N93Q/edit#gid=0 <https://docs.google.com/spreadsheets/d/18lJFzTITGgSfx1OGH_KkDWdKXgkGTt5et7Lv4X5N93Q/edit#gid=0>

2. There is a top down vehicle view in the slide deck, but I put it here as well:
https://docs.google.com/drawings/d/1ByPhEoScWatuC8Mcl9LLpOba4My0Iv3-eyuXoiz7svQ/edit

dcmcshan, 2015-05-25:
Oh, I see your quesiton about Mission Duration.  Ultimately, there will be a “Mission W/kg” concept of how much energy it takes to fly, but we don’t have that yet…  So for now, lets do this: let’s add a Mission W/kg that is 2x what the Hover W/kg is and computer the time off that.

I updated the Mission Plannning spreadsheet accordingly.

dcmcshan, 2015-05-25:
While I see the FuelGauge header and source, it doesn't seem to show up in my project... any ideas?

BojanKverh, 2015-05-25:
Daniel, sorry, I was busy working on other gauges and didn't see this problem. Looks like MissionPlanner.pro didn't get updates. I will fix it.

BojanKverh, 2015-05-25:
Ok, now this issue should be fixed

dcmcshan, 2015-05-25:
While I now see the other headers and source in the project, they are not in the MissionPlanner widget.

BojanKverh, 2015-05-25:
That's strange, if I go to DroneWidgets/MissionPlanner folder, all files are there. Except QcGaugeWidget, which is based in another folder, which you uploaded yesterday.

dcmcshan, 2015-05-25:
Is what my world looks like when I run MissionPlanner

BojanKverh, 2015-05-25:
Hm, I apologize, but I don't really understand what is the problem then. Worst case, I can send you .zip file again over Upwork message system.

dcmcshan, 2015-05-25:
I’m not a Qt expert, but should I expect to see FuelGauge instantiated within MissionPlannerDataWidget?

MissionPlannerGauges.h and cpp seem to be empty...

BojanKverh, 2015-05-25:
Uh, you're right, sorry about that. Looks like I am having some trouble with Git. Will fix this issue and let you know.

BojanKverh, 2015-05-25:
Daniel, now it should be alright. It's been a while since I used git, so I forgot the a bit how to use it. I apologize for the trouble.

dcmcshan, 2015-05-25:
Much better!  Max Fuel is actually 6 gallons.  Max Weight is probably 100 kg

dcmcshan, 2015-05-25:
I feel Mission Planner is in sufficient prototype state for Tuesday.  I think move along to the Vehicle Widget!

BojanKverh, 2015-05-25:
Thanks for your review, Dan. I edited the Max Fuel and Max weight limits according to your suggestions.

closed 2015-05-26 by dcmcshan