Syzygyx/DroneWidgets #5: Vehicle widget

BojanKverh, 2015-05-25:
Vehicle widget code is also uploaded now. I created a new folder/application for it. The provided .svg file of the vehicle's top view is not really a vector image, it is a pixmap converted into svg format, so it doesn't scale very well. It is also not possible to detect the right positions for the gauges, so I have to find it by clicking on the centers and outputing the position of the click. I believe this could be done with the right .svg vector file. VehicleWidget can take many different gauges and places them at different positions and each gauge can have its own current value. Right now, they are all connected to the same value slider, but this can be changed arbitrarily. Dan, please check if this is what the widget was supposed to do.

BojanKverh, 2015-05-25:
I will finish working for today. If there are still any issues to fix, I will do it tomorrow.

dcmcshan, 2015-05-26:
> itHub <https://github.com/Syzygyx/DroneWidgets/issues/5#issuecomment-105333556>.

Would seem there is no VehicleWidget.h in the repository!

dcmcshan, 2015-05-26:
Files are there, but something seems to be amiss with dependencies on the gauge library?

dcmcshan, 2015-05-26:
Added sources from the gauge directory.
Added +svg 

Now I get below.  Assuming there is some sort of pathing issue.

Cannot open file ':/files/Vehicle.svg', because: Unknown error

dcmcshan, 2015-05-26:
Assuming you can easily sort out the building issues, the only other thing I would ask is 7 sliders on the left to change the gauge values for prototype demoing…

Thanks!  I think we’re pretty good!

dcmcshan, 2015-05-26:
Also, please make backround black/dark grey.

Thanks!

BojanKverh, 2015-05-26:
I created 7 sliders and put it to the left of the Vehicle's widget and I also changed the background color. Regarding missing file, I check both Files.qrc and files/Vehicle.svg files and both seems to be ok. Files.qrc should be just a tiny XML file, which Vehicle.svg should be a 1.2MB XML containing mostly coded binary data. Just in case you are missing it, I sent you Vehicle.svg over the Upwork message system. If there is still this issue, please let me know.

dcmcshan, 2015-05-26:
maybe send me zip file?

BojanKverh, 2015-05-26:
I sent you the zip file of VehicleWidget through Upwork message system.

closed 2015-05-26 by dcmcshan