Syzygyx/DroneWidgets #6: Integrate with QGC

dcmcshan, 2015-05-26:
The standalone widgets need to be integrated with QGroundControl to show up in the Tool Widgets menu.

BojanKverh, 2015-05-26:
OK, I will first solve the problem with OGC compiling and then start working on this issue.

BojanKverh, 2015-05-26:
I was able to compile OGC, will look around it and its source code a bit.

BojanKverh, 2015-05-26:
OK, I got an idea about how this software is built and how to integrate a new widget into the Tool Widgets menu. I've got a few questions:
1. Which widget would you like to see integrated first?
2. How are we going to share the source code? Do you intend to clone the QGroundControl source tree and have it somewhere inside the DroneWidgets?

dcmcshan, 2015-05-26:
1. Let’s integrate the VehicleWidget first
2. I think the right answer is that QGroundControl wants to be a separate source tree and we just add the widgets to that.  So we can update QGC as it does.  Right now it’s under DroneWidgets as a submodule, I think...

BojanKverh, 2015-05-26:
But do we want to add files directly to the QGroundControl main source tree? Yes, QGroundControl is now a submodule, but as far as I understand, it is a direct link to the main source tree or I am missing something?

dcmcshan, 2015-05-26:
I don’t think you can add files to a submodule as such… can you?  I don’t know.  Anyways, if there is a way to keep the widgets separate from the QGC source tree, might be best for now.

BojanKverh, 2015-05-26:
OK, let me rephrase the question: I have a copy of QGroundControl on my laptop, I will do the changes and integration here. How will we then transfer the changes to your computer for you to check them with minimal effort?

BojanKverh, 2015-05-26:
I made the VehicleWidget scalable, but it preserves the ratio between width and height, otherwise the gauges wouldn't fit (they have to be circles). I also integrated the new scalable widget into the QGC. I think I only edited 3 files, should I send them to you as zip for starters?

dcmcshan, 2015-05-26:
Sure, that works.

BojanKverh, 2015-05-26:
OK, file sent through Upwork message system.

dcmcshan, 2015-05-26:
Takes me a long time to build all of QGC.  Is there some way to use the “Load Custom Widget File” mechanism under the Tool Widgets menu?  Would be nice to be able to just build the widgets separately…  At least for now?  Or is that truly for their internal custom widgets?

dcmcshan, 2015-05-26:
Also, I don’t think that zip approach is sufficient.  This warning below tells me it’s looking for that qcgaugewidget project.  I think the answer needs to be either a widget that can be compiled separately, or you will just have to edit the qgroundcontrol directory in the DroneWidgets directly for now (but shouldn’t have dependencies outside the qgroundcontrol directory - e.g. qcgaugewidget).

WARNING: Failure to find: src/ui/qcgaugewidget.h

dcmcshan, 2015-05-26:
In the future, if you can make it a practice to do a fresh checkout from git and build on your side, I think it might save us a lot of time. Thanks!

BojanKverh, 2015-05-27:
Dan, I agree that this approach is very bad and we definitely have to think of better way of sharing the files.  The best way would indeed be if I can compile the widget as some sort of plugin and then load it dynamically into QGC. The problem is, that I don't have the "Load Custom Widget File" in the Advanced/Tool widgets menu.

dcmcshan, 2015-05-27:
This is what my Tool Widgets menu looks like.  

In any event, for now lets not worry about the plugin concept.  What I would like you to do is to create a copy of the qgroundcontrol source in the repository (rather than a submodule), and make your modifications in that.  Again, all changes should be inside that directory, so you’ll likely need to add that custom gauge source within.  Please checkout fresh and build to confirm.

Thanks!

BojanKverh, 2015-05-27:
OK, I created a copy of QGC source code tree, I named it qgroundcontrol-fork.
The only difference to the original source code tree is that when creating Makefile with qmake, you have to specify a file, so you execute "qmake qgroundcontrol.pro" instead of just "qmake", because the .pro file name now differs from the folder (I didn't want to rename anything in order not to break some dependencies). I will add VehicleWidget and all the dependencies into this source code tree as well.

dcmcshan, 2015-05-27:
Ah, okay.  I was just in process of removing the submodule entirely.  So, you may see that qgroundcontrol is no longer there.  Apologies if that is confusing.

BojanKverh, 2015-05-27:
No problem. Maybe sometime in the future we can rename it back to qgroundcontrol, but I suggest that I first try to make integration, so that we see that the concept works.

BojanKverh, 2015-05-27:
OK, I made the integration and checked the GitHub archive and everything seems to be in place in the fork folder. So please try to compile it and see if it works.

dcmcshan, 2015-05-27:
What should we do about the mavlink submodule?  It seems to not work for me...

$ git submodule init
No submodule mapping found in .gitmodules for path 'qgroundcontrol-fork/libs/mavlink/include/mavlink/v1.0’

dcmcshan, 2015-05-27:
Again, I would request that you check out from git and try to build on your side to make sure everything works.  I suspect you will see same issues I do.

dcmcshan, 2015-05-27:
I reverted my previous deletion of qgc in hopes it might help with the submodule.  It did not.

BojanKverh, 2015-05-27:
git checkout did nothing, unfortunately

BojanKverh, 2015-05-27:
Moved the entire folder to some other location, and cloned the repository again, I see the issues you described indeed. But no idea how to solve them

BojanKverh, 2015-05-27:
OK, I got an idea reading through some online help, will try to solve this.

dcmcshan, 2015-05-27:
I think the way you are supposed to do it is to actually git fork the QGC library rather than copying it.  I get the sense that git repos do not like to be copied...

BojanKverh, 2015-05-27:
Yes, I have the same feeling :D

dcmcshan, 2015-05-27:
I am actually going to be travelling next couple of days, so I won’t be messing with the repository at all. 

Once you have something that directly builds, let me know and I’ll start the build - it takes a while on this laptop...

BojanKverh, 2015-05-27:
OK, I will try to resolve this issue as soon as possible.

BojanKverh, 2015-05-27:
Dan,
one possible solution that I found is to copy the mavlink files brute-force into the right folder and then compile the application. With Linux command line you need to do this (if you are in the DroneWidgets main folder):
1. cd qgroundcontrol-fork/libs/mavlink/include/mavlink/v1.0/
2. git clone --recursive https://github.com/mavlink/c_library
3. mv c_library/* .
4. rm -rf c_library/
5. cd ../../../../..
6. qmake qgroundcontrol.pro
7. make -j9

Comments:
1. go to the right folder
2. copy the mavlink files (this will create a c_library folder in qgroundcontrol-fork/libs/mavlink/include/mavlink/v1.0/
3. move all files and subfolders from c_library folder one level up (to qgroundcontrol-fork/libs/mavlink/include/mavlink/v1.0/)
4. remove c_library folder (optional)
5. go back to qgroundcontrol-fork folder
6. Run qmake on qgroundcontrol.pro
7. Compile the application

I deleted the entire DroneWidgets folder and pulled it down again and by doing these 7 steps I was able to compile the entire app. I hope this helps.

BojanKverh, 2015-05-27:
This is how it looks like:
[image]

dcmcshan, 2015-05-27:
Looks nice :)

dcmcshan, 2015-05-27:
Too complicated.  I forked qgroundcontrol entirely and created a new repository:

https://github.com/Syzygyx/qgroundcontrol

I have added you to this repository.  In theory you can make the needed changes in their, and the mavlink submodule will still update correctly.

I think that’s maybe the “right way”?

BojanKverh, 2015-05-27:
OK, so I should pull the Syzygyx/qgroundcontrol source code tree and make the integration there?

dcmcshan, 2015-05-27:
for now, yes.

apparently you can’t make repository forks private, so probably not our long term solution

BojanKverh, 2015-05-27:
OK, now I managed to upload all the files, please check if I got it right this time.

dcmcshan, 2015-05-27:
Building now!  Repository wise looks happy.

BojanKverh, 2015-05-27:
Great :)

BojanKverh, 2015-05-28:
Can we close this issue?

closed 2015-06-04 by dcmcshan