Syzygyx/DroneWidgets #7: Advanced QGC Integration

dcmcshan, 2015-05-26:
This is where we "wire up" the gauges in QGC to MAVlink parameters.

BojanKverh, 2015-05-27:
One question regarding these issue: do we need any other gauges except the Temperature ones?

dcmcshan, 2015-05-27:
Yes, just added to queue!

BojanKverh, 2015-05-27:
Will look into integrating MAVlink parameters and gauges tomorrow.

BojanKverh, 2015-06-04:
Just a quick question before I can continue the work: it looks to me that the application gets the parameter values like speed, altitude, etc. from an object of type MAVLinkDecoder, do you know if that is correct?

dcmcshan, 2015-06-04:
That might be so.  I’m not super familiar.  The way this works is that MAVLink flows into the app via serial port or socket.  It’s an XML stream containing parameter data.  I’d sorta expect that QGC might store variables internally, but maybe not…

BojanKverh, 2015-06-04:
I've connected the ground speed message to the AirspeedIndicator and FlightInstrumentsWidget. Since I have now way to check if this works right now, I would kindly ask you to test it and report. If this works, the integration of the rest of the parameters should be quite straightforward once I figure out the name of messages corresponding to the other values.

dcmcshan, 2015-06-05:
Here’s a log file to read in.  I just found it on the web, so no idea what it actually is…

closed 2015-06-08 by dcmcshan