Syzygyx/DroneWidgets #10: Create telemetry log file

dcmcshan, 2015-06-04:
Create a log file of telemetry values for playback in QGC to demonstrate gauge and visualizer actions

dcmcshan, 2015-06-05:
I have added a telemetry log file (log.mavlink) to this repository.  I just downloaded it off the web, so have no idea what's in it, but I can load it in qgc and it seems to work...

dcmcshan, 2015-06-05:
It's not obvious to me that QGroundControl actually updates the widgets while replaying the log file. . Might be something you have to force it to do...

BojanKverh, 2015-06-05:
Either that, or the file format is outdated. After loading the file, I get lots of these messages written in the console:
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 246 )
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 247 )
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 248 )
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 249 )
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 250 )
ParameterLoaderLog: Read re-request for (componentId: 1 paramIndex: 251 )

dcmcshan, 2015-06-05:
If you go to the “Analyze” tab, you can see values updating.  I can see airspeed updating, for instance.  But the Primary Flight Display does not reflect this, so I suspect there is a hook we have to call in the replay to update the parameters.

BojanKverh, 2015-06-05:
Good observation, I will check that.

BojanKverh, 2015-06-05:
I figured out how to get the airspeed to show: first update to the latest version on GitHub and compile. Then run and add the Unmanned systems from the Tool Widgets menu. Then load the log file and two systems will appear in the Unmanned systems box. Click with right button on the MAV 001 box and select the "Control this system" option. From the beginning of the log file speeds are very low, so you will not see the difference in the Airspeed indicator, but if you move the file time slider to 08:45, you will notice that the correct speed is shown (in knots).

BojanKverh, 2015-06-05:
Integrated altitude and heading

dcmcshan, 2015-06-05:
Looks good.

BojanKverh, 2015-06-05:
Integrated also vertical speed, so basically 4 indicators (airspeed, altitude, heading and vertical speed) should now work. My question is how are we going to update the fuel gauge? By approximating the fuel quantity by subtracting time of flight multiplied by assumed fuel consumption per time unit, or is there a sensor, which reports fuel quantity in these systems? I guess weight gauge will also use this information to decrease the total vehicle weight over time.

dcmcshan, 2015-06-05:
There will be sensors, and some math.  For now, isn't critical.

BojanKverh, 2015-06-05:
OK then, so I guess we leave those two gauges as they are for now. Is there anything else I can start working on now? You mentioned the FlightGear simulator integration?

closed 2015-06-08 by dcmcshan