Syzygyx/DroneWidgets #11: Windows Executable

dcmcshan, 2015-06-04:
Can you make me a windows executable (.exe) of our latest baseline.  Customer would like to evaluate, and I have no windows machine!

BojanKverh, 2015-06-04:
Uh, unfortunately I also don't have Windows computer right now. I could only borrow it from my colleague on Monday.

dcmcshan, 2015-06-05:
If you can get me a windows executable Monday, that would be helpful.  Thanks!

BojanKverh, 2015-06-05:
I will do my best. I am thinking about doing it in Windows virtual machine on Linux, I've done that before, but I don't remember the details

BojanKverh, 2015-06-08:
OK, I installed Windows virtual machine on my laptop yesterday and will try to do this first, since the deadline is today.

BojanKverh, 2015-06-08:
Unfortunately, I am stuck with 5 error messages similar to this one:
XbeeLink.obj:-1: error: LNK2019: unresolved external symbol xbee_setupAPI referenced in function "private: bool __cdecl XbeeLink::hardwareConnect(void)" (?hardwareConnect@XbeeLink@@AEAA_NXZ)

I manually included the xbee library into the .pro file, but to no success. Do you have any idea how to solve this? Because everything else seems to compile fine, it's just this problem.

dcmcshan, 2015-06-08:
> XbeeLink.obj error: LNK2019: unresolved external symbol xbee_setupAPI referenced in function "private: bool __cdecl XbeeLink::hardwareConnect(void)" (?hardwareConnect@XbeeLink@@AEAA_NXZ)

Unfortunately, I don’t know.

You should be able to configure Xbee off.  We don’t need it. 

No mention of anything in the build instructions, either.
https://github.com/mavlink/qgroundcontrol/blob/master/README.md

BojanKverh, 2015-06-08:
OK, I will try that.

BojanKverh, 2015-06-08:
Thanks for the tip, it's indeed possible to turn the Xbee off, so now the application compiles well. The next problem is that it crashes soon after start up.

dcmcshan, 2015-06-08:
I hate windows...

BojanKverh, 2015-06-08:
Me too :D I will try to recompile everything from scratch.

BojanKverh, 2015-06-08:
Same thing :( Debugger reports Access violation in opengl32.dll

dcmcshan, 2015-06-08:
What windows are you running?  We are looking to target Windows 7.

Here are some instruction from a previous developer on this project for compiling QGC under windowWe  ar 

1) Install Qt Creator 3.3.0 / Qt54Msvc32bitOpenGL (Based on Qt 5.4.0 MSVC 2010, 32bit, Build on Dec 8 2014 at 15:24:21)
1.1) Add qmake.exe path to environment variables
1.2) C:\~\Qt54Msvc32bitOpenGL\5.4\msvc2013_opengl\bin
2) Install Microsoft Visual Studio Community Edition 2013, Version 12.0.31101.00 Update 4
3) Download qgroundcontrol
4) Open Qt Creator
5) Open qgroundcontrol-master\QGCApplication.pro in Qt Creator
6) Edit / Add files to QGC via Qt Creator as needed
7) Generate Microsoft solution file from command line (needs to be done each time files added via Qt Creator)
7.1) C:\~\qgroundcontrol-master>qmake -r tp vc qgroundcontrol.pro
7.2) 3 files should be generated: 
  1) QGCApplication.pro
  2) QGeoServiceProviderFactoryQGC.vcxproj 
  3) qgroundcontrol.vcxproj
8) Open Microsoft Visual Studio
9) Open newly created Microsoft solution file QGCApplication.pro in Microsoft Visual Studio
9.1) 2 subprojects are automatically loaded
  1) QGeoServiceProviderFactoryQGC 
  2) qgroundcontrol
10) Build QGC from Microsoft solution subproject qgroundcontrol.vcxproj
10.1) Right click qgroundcontrol -> click Build
10.2) Initial build / after clean takes ~ 10 minutes
10.3) Confirm build succeeded in Microsoft Visual Studio Output window: ========== Build: 2 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
10.4) Confirm new build product / exe C:~\qgroundcontrol-master\debug\qgroundcontrol.exe

> On Jun 8, 2015, at 2:51 PM, BojanKverh <notifications@github.com> wrote:
> 
> Access violation in opengl32.dll

BojanKverh, 2015-06-08:
I am also using Windows 7. I compiled this app with Qt Creator, will try this procedure as well.

BojanKverh, 2015-06-08:
The same problem remains even after compiling it with VS.

dcmcshan, 2015-06-08:
I wonder if it’s something to do with the virtual machine?

dcmcshan, 2015-06-08:
I see that qgroundcontrol is only supported for 32-bit Windows?  Are you running 64 bit?

dcmcshan, 2015-06-08:
Also, relevant to the OpenGL issue, the build instructions say to not include opengl in Qt.
Install Qt5.4

Download Qt 5.4 from here: http://download.qt-project.org/official_releases/qt/5.4/5.4.0/qt-opensource-windows-x86-msvc2013_opengl-5.4.0.exe <http://download.qt-project.org/official_releases/qt/5.4/5.4.0/qt-opensource-windows-x86-msvc2013_opengl-5.4.0.exe>
The Qt variant should be for VS 2013, 32 bit (not 64) and include opengl.

BojanKverh, 2015-06-08:
Uh, I missed that part about 32 bit.

BojanKverh, 2015-06-09:
Still getting the same error, must be that something is wrong with virtual machine indeed.

dcmcshan, 2015-06-09:
Did you use the specified QT (without opengl) or the one that comes with QT Creator?

BojanKverh, 2015-06-09:
I installed qt from your link

dcmcshan, 2015-06-09:
Very strange.  Did you say you had a friends windows machine you could borrow?  This is fairly critical as customer’s machine is a win7 laptop!

BojanKverh, 2015-06-09:
Yes, but I couldn't borrow it, because he needs it these days. I will continue to search for solution.

BojanKverh, 2015-06-09:
OK, I read now that there is an issue with Virtual Box and OpenGL. I found a company from which I can borrow Windows laptops, I will call them tomorrow.

dcmcshan, 2015-06-09:
Cool.

BojanKverh, 2015-06-09:
I've got a Windows 7 laptop now, will try to compile QGC.

dcmcshan, 2015-06-09:
Good luck!

BojanKverh, 2015-06-09:
Thanks, I will need it :)

BojanKverh, 2015-06-09:
OK, looks like I did it. :) Now I will try to pack everything into a nice installation package, so that you don't have to manually copy the app and dlls.

BojanKverh, 2015-06-09:
I added the self extracting installation file (QGroundControl.EXE) to the qgroundcontrol archive (maybe not the best idea, you can delete it after downloading it), so please check if it runs correctly. The first time I ran it, it crashed (I guess because of some missing config file), but every other time it ran perfectly.

BojanKverh, 2015-06-09:
It still doesn't work because Iexpress can't handle subfolders (facepalm). Will have to look for better installation app.

BojanKverh, 2015-06-09:
OK, now I managed to get all files included. But the self extracting file is quite big (75MB), which is over the recommended GitHub limit, so maybe we should share it over some other server?

dcmcshan, 2015-06-09:
Dropbox?

dcmcshan, 2015-06-09:
Can we include flightgear in the (very large) installer?

BojanKverh, 2015-06-09:
Dropbox would be ok I think. I guess we could also include flightgear (I found that compiled windows version is around 1GB)

BojanKverh, 2015-06-09:
Oh, and how often do you think we should do the QGC Windows compile?

dcmcshan, 2015-06-09:
As infrequently as possible... only for customer delivery

BojanKverh, 2015-06-10:
OK.

dcmcshan, 2015-06-10:
What's a good email for you for dropbox?  I will share a dropbox folder to keep the Windows executable in...

BojanKverh, 2015-06-10:
bojan.kverh@gmail.com

dcmcshan, 2015-06-11:
Think you can get me a tested windows installer today?  Thanks!

BojanKverh, 2015-06-11:
I think yes, I will compile the latest commit. Would you like to have FlightGear package included?

dcmcshan, 2015-06-11:
No, let’s not include it at the moment.  Make sure that it can find the default install (I’m not convinced that “fgfs.exe" is on the path).

We will make a custom install of flightgear that includes the vehicle model we need.

BojanKverh, 2015-06-11:
Is there a place on DropBox, where I can upload it?

dcmcshan, 2015-06-11:
I shared a folder with you yesterday, I think?

BojanKverh, 2015-06-11:
Strange, because I don't seem to see it in DropBox.

dcmcshan, 2015-06-11:
https://www.dropbox.com/sh/3trisf4umm3fod2/AAA8z1VntAJzTke-8_lHXOqAa?dl=0

Try that

BojanKverh, 2015-06-11:
I can open that, but it doesn't seem I have any rights to upload the file there.

BojanKverh, 2015-06-11:
Now it's ok, I got your invitation and uploaded the Windows installation package. I compiled it a few hours ago, so it doesn't have all the improvements I did today, but  I can compile it again, if necessary.

dcmcshan, 2015-06-11:
Have you tested it on a fresh install (maybe your virtual machine) to make sure we don't have any dll or link or path issues?  If not, I'll take a look tomorrow before I send to customer.

BojanKverh, 2015-06-11:
I did it last time, and it didn't report any missing DLL (but it crashed because of that unresolved OpenGL/virtual box issue). This time I did not, but I can do it to make sure.

dcmcshan, 2015-06-11:
Thanks!  I am going to send to customer and want to be sure it works with the installer :)

BojanKverh, 2015-06-12:
I checked on a fresh virtualbox and it runs until the OpenGL issue, so it should be ok. It installs in C:\Program Files (x86)\rtGCS folder. Sorry because I wasn't able to make the installer create a Desktop icon yet (I will look into it until next build), pointing to the application. I hope this won't be the issue with the customer.

dcmcshan, 2015-06-12:
Okay, if you get around to making the start menu / desktop icon, that will be good.  In the meantime, i can just tell them to run from C:\ProgramFiles

dcmcshan, 2015-06-12:
Have you checked out flightgear integration on windows?  I'm worried about path...

closed 2015-06-12 by dcmcshan