Syzygyx/DroneWidgets #12: Integrate FlightGear as Visualizer

dcmcshan, 2015-06-05:
Goal is to use FlightGear (http://www.flightgear.org/) as a visualizer for QGC. We already have a vehicle model which I will get to you.  For now, you really just have to focus on having QGC output position and attitude to FlightGear.  Later, we'll have it send other telemetry as well (like motor speed so the props spin...)  For now, just focus on ability for QGC to send FlightGear 6DOF position/attitude

BojanKverh, 2015-06-05:
I studied the FG protocol of data exchange, will try to make a working prototype tomorrow.
One question though: FG has the ability to read data from file and replay it, so is this the option too, or QGC has to communicate with FG in real time (not that it makes much difference, IMHO)?

dcmcshan, 2015-06-05:
Realtime.  We want FG to reflect what is in QGC

BojanKverh, 2015-06-06:
Created a small generic protocol description file (xml/qgc.xml) to use when communicating with FlightGear. When outputing data from FG, this works fine. Inputting does not and documentation about this particular issue is rubbish, so I am stuck for the moment.

BojanKverh, 2015-06-06:
OK, I figured out how to send parameters from QGC to FlightGear. FG expects data in network order (which is Big Endian), while PC processors are usully Little Endian.

dcmcshan, 2015-06-06:
Yes... silly wrong endian Intel.  I've been designing embedded systems for over twenty years and it still seems ridiculous to me.

BojanKverh, 2015-06-06:
Now it is possible to replay QGC log files like log.mavlink with FlightGear. Two minor issues though:
- heading instrument at FG always shows +90 in regards to QGC, even though QGC sends it the correct value (I am not a pilot but AFAIK yaw=0 means north, while FG will point to the east in this case)
- altitude meter: this one confuses me, because the documentation clearly states that altitude should be sent in meters, although to me it looks like it requires altitude in feet and its instrument shows it in meters...

dcmcshan, 2015-06-07:
I added "arducopter" directory to the DroneWidgets repo.  You can drop this in your custom airframe folder and it will render the vehicle.  

Do you think you can create a widget for QGC that launches flightgear with the needed arguments to listen to the udp port?  I believe there is already a "simulator" widget which will start flight gear with the airframe, but it needs different parameters to do what we want it to do...

BojanKverh, 2015-06-07:
Yes, I believe it can be done.

BojanKverh, 2015-06-07:
Launching FG can now be done from QGC (Advanced/Launch FlightGear)

dcmcshan, 2015-06-07:
Need to be able to set path. FlightGear is installed (I am on a mac…)

BojanKverh, 2015-06-07:
OK, I guess "Launch FlightGear" should open a dialog, where you can select the executable (with its path) and possibly an aircraft, as I guess this will not be used only with arducopter?

dcmcshan, 2015-06-07:
Are you using QGCFlightGearLink::connectSimulation?

It seems to have a bunch of logic for sorting out flightgear launch.

For now, we can default to arducopter

BojanKverh, 2015-06-07:
No, I did not use that, because I didn't noticed it. Will take a look into it.

dcmcshan, 2015-06-07:
Of course, it seems that the default HIL Simulation doesn’t actually work on Mac anyways… I will investigate a bit.

dcmcshan, 2015-06-07:
Launcher aside,

If I do this from the /Applications/FlightGear.app/Contents/MacOS directory, flightgear starts with our model (I copied it into the Aircraft directory)

./fgfs --aircraft=arducopter --native-fdm=socket,in,100,,5600,udp --fdm=null

But seems not to be linked to QGC when I playback.  Is there something else I should have to do?

BojanKverh, 2015-06-07:
As far as I understood Mac OS X, the FG executable should be  /Applications/FlightGear.app
That's how I now updated the launcher. Please check if it works.

dcmcshan, 2015-06-07:
That is the application, but if you want to pass arguments, the executable is actually as I mentioned,

/Applications/FlightGear.app/Contents/MacOS/fgfs 

Mac OSX is essentially unix.

BojanKverh, 2015-06-07:
OK, thanks for that information

BojanKverh, 2015-06-07:
One remark though: if QGC starts sending flight data to FG from the beginning, the FG seems to stay stuck with the Loading scenery message forever. Since there seems to be no precise way to determine, when it is safe to start sending flight data, I implemented a timeout of 20 seconds, which seem to work. So FG will start receiving flight data from QGC only 20 seconds after the FG was launched.

dcmcshan, 2015-06-07:
Looking through the code, I don’t see how you are sending data to flightgear?  Are you using 
MessageDispatcher::SendUDP()?

It seems to have everything commented out…

BojanKverh, 2015-06-07:
Yes, through SendUDP. The code which is commented out was used in the beginning, when I tried to establish the communication with FG. I just wanted to make sure that this simple example works. Later, I left it in, so that I can look back into it in case of any troubles. But after the commented out code, there are few lines, which actually send the internal m_netFDM variable through UDP.

dcmcshan, 2015-06-07:
I have it up and running, though it appears that flight gear is rendering it underground.  I suspect this is because you are sending it AMSL which is altitude above mean sea level, and flight gear is perhaps expecting AGL (above ground level)

There are several reasons why this might occur, but either way we might should bias altitude so that altitude is somewhat reasonable.

BojanKverh, 2015-06-07:
I am sending AMSL, because in net_fdm.hxx it is clearly stated, that altitude should be given in meters above sea level. There is another variable there, agl, which is altitude above ground level, but to calculate that, I would need to know the terrain elevation at current position. UAS doesn't tell AGL as far as I know. It does tell relative altitude, which is altitude above the home point elevation. I took this one as AGL for the moment.

closed 2015-06-08 by dcmcshan