Syzygyx/DroneWidgets #16: Attach servo output to vehicle gauges

dcmcshan, 2015-06-08:
For illustration purposes, connect servo1_raw - servo4_raw to our gauges in vehicle widget and scale appropriately.  Obviously as the data is only 4 servos, and we have 7 gauges, you will need to use some more than once.  It doesn't much matter, I just want for illustration.

BojanKverh, 2015-06-10:
OK, I prepared the basis communication, but I will need some help, because I am not sure what kind of text messages MAVLinkDecoder will send out containing these servoX_raw values. MAVLinkDecoder will now write SERVO_MSG: + message to the standard output, whenever it has to emit these message. Is it possible to compile the latest commit and run it with real system and copy/paste me the output lines beginning with SERVO_MSG?

dcmcshan, 2015-06-10:
Unfortunately, we don’t have a real system.  However, the log file we were using is populating the four servos, so we should see that…

dcmcshan, 2015-06-10:
servo1_raw - servo4_raw are active in our log file.  They "zero" at 1131 and "max" at about 1650.   scale accordingly.  if you can also update the digital readout with the raw value, that might be cool as well.  again is just for demonstration.

dcmcshan, 2015-06-10:
FWIW, it’s not printing out values:  (but see other messages - we just want to wire up servo1-4 for now)

SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo1_raw"
SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo2_raw"
SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo3_raw"
SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo4_raw"
SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo5_raw"
SERVO-MSG: "port0_SERVO_OUTPUT_RAW.servo6_raw"
SERVO-MSG: “port0_SERVO_OUTPUT_RAW.servo7_raw”

BojanKverh, 2015-06-10:
Good observation, thanks, this is what I needed.

BojanKverh, 2015-06-10:
This is now done with the latest commit, I connected both, temperature and RPM with different formulas to servo_raw values.

closed 2015-06-12 by dcmcshan