Syzygyx/DroneWidgets #19: Payload Waypoint

dcmcshan, 2015-06-09:
In the Mission Plan widget, add waypoints type: "Payload 1", "Payload 2", and "Payload 3"

BojanKverh, 2015-06-09:
Could you please be more specific about this?

dcmcshan, 2015-06-09:
On this screenshot, 

http://qgroundcontrol.org/_detail/users/qgc-v1.0.0-operator-3d-map1.png?id=screenshots

the first field in the bottom widget reads NAV: Waypoint.  We simply want one that reads "Payload 1/2/3".  It's to define a "mission" in which some payload activity needs to take place.  Perhaps a more consistent approach is to make it "PAY: Deploy 1" "PAY: Deploy 2" "PAY: Deploy 3" in case we have other activities we want to define later...

BojanKverh, 2015-06-10:
Thanks for the explanation. I did this and also when you move the waypoint up, down or delete it, the texts "PAY: Deploy" are renumbered again, so that the first waypoint in the list always has "PAY: Deploy 1" text, I hope that is correct. I noticed also, that waypoints background color doesn't get updated correctly, if you move/delete WP, I guess I should fix this too?
I will continue tomorrow.

dcmcshan, 2015-06-10:
Payload numbers should remain static.  Imagine you had three different payloads that you were going to release over three different locations.

BojanKverh, 2015-06-10:
OK, I will just remove the renumbering then.

dcmcshan, 2015-06-10:
These are still renumbering, so I moved back to "In Progress"

BojanKverh, 2015-06-10:
OK, then we didn't understand each other well. 
If I add 4 waypoints into the mission, the plan will be:
PAY: Deploy 1
PAY: Deploy 2
PAY: Deploy 3
PAY: Deploy 4

Now if at PAY: Deploy 3, I click to move it up, the plan will be:
PAY: Deploy 1
PAY: Deploy 3
PAY: Deploy 2
PAY: Deploy 4

How it should look like then to make it right?
And how it should look like, if I choose to remove PAY: Deploy 2 in the original mission?

dcmcshan, 2015-06-10:
So, there should be only 3 entries in the list, all static.  "PAY: Deploy 1", "PAY: Deploy 2" and "PAY Deploy 3".  There are three payloads on the vehicle, and we can “deploy” them.  The index is simply part of the label, and isn’t ever dynamic. Moving them in the list doesn’t change which payload we are deploying.  

Sorry for the confusion.

BojanKverh, 2015-06-10:
No problem. So basically there should be 3 fixed entries, without a possibility to move them around, remove them from the list or add new ones to the list?

closed 2015-06-12 by dcmcshan