Syzygyx/DroneWidgets #20: Hack Mission Plan to Allow Mission Definition without Attached Vehicle

dcmcshan, 2015-06-09:
By default, you can't point and click waypoints on the may without a system connected.  Not sure why, but for some upcoming tasks, I'd like to be able to define a mission without anything connected...

BojanKverh, 2015-06-09:
Do you mean on the Plan, when you right click on the map and select "Go to here" or "Set home locaion here"?

dcmcshan, 2015-06-09:
Right, when vehicle is attached it will actually let you view and edit waypoints on the map.  Not sure why.

This is helpful to accomplish because one of the next things is going to be to define "geofence" polygons on that map, and I don't want you to have to connect a vehicle just to test it.

BojanKverh, 2015-06-09:
I did this with the latest commit, however only the Home location can be set. "Go to here" options also require UASWaypointManager to be present, which I guess will not be true until UAS is connected indeed.

dcmcshan, 2015-06-09:
Can we add UASWaypointManager without it?

Basically we are looking to be able to design a mission with waypoints and boundaries without needing a vehicle attached...

BojanKverh, 2015-06-10:
I guess it must be possible, I will look into it.

BojanKverh, 2015-06-10:
I created a default UASWaypointManager and assigned it a fake UAS object (without that it will crash, because on several places in the code, the manager addresses the UAS without checking whether it is 0 or not). Now 'Go to Here' option set the selected WP in this default manager, however nothing happens on the map visually. In the UASWaypointManager::goToWaypoint method, there is this code later on:
        mavlink_msg_mission_item_encode(uas->mavlink->getSystemId(), uas->mavlink->getComponentId(), &message, &mission);
        uas->sendMessage(message);
which I suspect does something, that should reflect on the map, but I am not sure how to do this with our fake UAS so far.

dcmcshan, 2015-06-10:
I found this MockLink connection under the Connect menu top right.  Maybe that does what we need?

BojanKverh, 2015-06-10:
I will look into it.

BojanKverh, 2015-06-10:
Well, it looks like you can add links that communicate through serial port, UDP or TCP socket, but nothing changes on the map after I connect to one of those.

BojanKverh, 2015-06-10:
With the latest commit, creating waypoints on the map by double-clicking on the location is now possible without vehicle connected. Not sure what happens when a vehicle is then connected on, if everything is still consistent.

closed 2015-06-12 by dcmcshan