Syzygyx/DroneWidgets #22: Change waypoint icon for payloads

dcmcshan, 2015-06-10:
So, you probably can't see this until you get the waypointmanager, but the map view should show the mission route - from waypoint to waypoint. We need to change the default waypoint icon for navigation waypoint to something that indicated payload waypoint.  Will need to be different color and different shape.  Maybe a red "upside down" teardrop?

BojanKverh, 2015-06-11:
Teardrop is now used as the waypoint icon

dcmcshan, 2015-06-11:
I haven't looked yet, but to confirm... Should just be for the payload type of waypoint to distinguish it.-

BojanKverh, 2015-06-11:
I see. So those "PAY: Deploy 1", "PAY: Deploy 2" and "PAY: Deploy 3" should actually be waypoint types and should be added to the drop down box, where there are already options "NAV: Takeoff", "NAV: Loiter unlim.", etc?

dcmcshan, 2015-06-11:
Correct!

BojanKverh, 2015-06-11:
OK, sorry that it took me so long to understand that.

dcmcshan, 2015-06-11:
My fault for not better specifying.  I'll try to do better in future.

BojanKverh, 2015-06-11:
It's ok, I hope this time I am closer to getting it right. :)

closed 2015-06-12 by dcmcshan