Syzygyx/DroneWidgets #26: Add Basic GeoFence Capability

dcmcshan, 2015-06-11:
A "GeoFence" is a region of airspace where the drone must not go - "no fly zone".  It's a polygon, defined by a min height and a max height and 3 or more lat/long points.  It's stored in a datafile as per this url:  https://pixhawk.org/users/geofence

For our needs, we want to be able to define the polygons on the map.   Ultimately it's pretty much exactly like a set of waypoints.   I'm not sure the best way to implement this, maybe we add a geofence tab to the waypoint window and when this tab is active, we make double click add points? We'd then want to be able to save and load the above referenced txt file

BojanKverh, 2015-06-11:
I understand. I guess there can be more than one "no fly zone" defined, so I guess every one of them must be saved in a separate file, according to the information on the link you provided? 
I agree about your suggestion, I think having a third tab "GeoFence" and when it is active, double clicking on the map should add another node to current "no fly zone" is the most intuitive way to do it.

dcmcshan, 2015-06-11:
Good question.  Not sure they get saved in multiple files, though.  I'll look into it.

BojanKverh, 2015-06-11:
Yea, it would seem unreasonable to save each one in separate file, but I haven't seen an example with multiple zones in one file either.

dcmcshan, 2015-06-11:
Lets assume they are all saved in the same file and are just separated by a blank line followed by the new polygon definition.

BojanKverh, 2015-06-11:
OK

BojanKverh, 2015-06-12:
I added GeoFenceZone class with basic information, which can be obtained from a text file. It doesn't do much yet, except read geofence.txt file and outputs one zone in both formats into out.txt file. I also added tab GeoFence in waypoint window. Will continue tomorrow.

BojanKverh, 2015-06-12:
I went through the waypoint window code now and haven't decided yet what approach to take. The GUI is done with Qt Designer (not Creator) where you design it by mouse and property editing. So I could either continue to add features to it in Qt Designer or try to add them by simply adding C++ code, which would generate the missing features. I frankly don't like Qt Designer, because I feel that by writing code I have more control and ultimately I am faster than by visual programming.

dcmcshan, 2015-06-12:
The most valuable feature we need right away is the ability to draw the polygon on the map and shade it in.  So, whatever gets you there fastest would be my vote.  We can refactor later.

BojanKverh, 2015-06-12:
Ok.

BojanKverh, 2015-06-13:
Now the app can draw the GeoFence zones read from "geofence.txt" file from the working directory. Unfortunately I lost some time with inheriting QGraphicsItem, I wanted to make the inherited class (GeoFenceZoneItem) refresh the same way as WayPointItem is refreshed, but it kept crashing the app, so I ultimately update it in a slot, which is connected to map->mapChanged signal in QGCMapWidget.

BojanKverh, 2015-06-13:
Loading and saving GeoFence files now works (GeoFence tab in Map View, at the bottom there are buttons "Save GFs" and "Load GFs"). Could you please explain how would you like the following to be done:
1. Listing the zones under GeoFence tab, how each zone (which can have arbitrary number of points) should appear there (we don't have zone names, we only have min and max altitude and list of points)
2. What should editing GF zone allow to user? Adding another node (point), deleting a node, moving a node, maybe moving entire GF zone?

dcmcshan, 2015-06-14:
I think the goal will be drawing the zones by hand on the map (imagine there is are areas you want to definitely avoid).  Being able to click and move the vertices of the polygon seems like something you should be able to do.  I think similar functionality to waypoint, but for geofence, if that tab is selected.

BojanKverh, 2015-06-14:
OK.

BojanKverh, 2015-06-14:
Now it is possible to load/save and edit (manually in the edit boxes, like in Waypoint tab) GeoFence zones. The edited changes do not yet reflect on a map, but they are saved in memory and if user chooses to save the zones into a file, the changes will be there. I will try to make the edited changes reflect on the map and adding points to current zone by double clicking on the map later today.

BojanKverh, 2015-06-14:
Edited changes in points longitude/latitude now reflect on the map. Adding points to current GeoFence zone is a bit ambiguous though; map widget is not necessarily "equipped" with mission plan widget (which consists of GeoFence tab also), so how do we decide if double click means new waypoint or new vertex in current GeoFence zone?

BojanKverh, 2015-06-15:
Clicking on GeoFence zone on the map makes it current (darkCyan background) in the Mission plan widget. Moving GeoFence zone points now works.

dcmcshan, 2015-06-15:
I think based on the tab in the waypoint window.  If it's on waypoint, double click adds waypoint.  If it's on geofence, we add a vertex.

BojanKverh, 2015-06-15:
OK, my point was that you don't necessarily have waypoint window there with map. You can turn it off. However, I think logic suggests that if waypoint window is present AND if it is set to GeoFence tab, then we should add vertex. Otherwise we add waypoint.

dcmcshan, 2015-06-15:
Ah, good point.  Maybe a radio button to select?

BojanKverh, 2015-06-15:
That's possible too, but where should we put it?

BojanKverh, 2015-06-15:
Now the implementation of GeoFence capability is as I suggested ("I think logic suggests that if waypoint window is present AND if it is set to GeoFence tab, then we should add vertex."), but adding can be easily changed to depend on radio button selection. 
Zone, which has darkCyan background, is the current zone. If there is a current zone, double clicking on map will add a vertex to current zone. If there is no current zone, double clicking will create a new zone and add the location of click to it as the first vertex. Clicking on WaypointList widget (and not on any zone editing widget), will make sure that no zone is current. 
Each zone also has a small button ("-") in top right corner, which removes it from the list and map.

dcmcshan, 2015-06-16:
Looks good, but vertices beyond 1 are 0.

dcmcshan, 2015-06-16:
They save to file correctly, though

dcmcshan, 2015-06-16:
And load from file correctly

BojanKverh, 2015-06-16:
Yes, I only forgot to update the input fields after adding point. Thanks for your report, I fixed this issue with the latest commit.