Syzygyx/DroneWidgets #27: Windows tGCS doesn't know where FlightGear is

dcmcshan, 2015-06-12:
Upon testing, I get message "Looks like fgfs.exe is not installed on this computer".  Even though it is.  I suspect it simply doesn't know where to look for it.

Mine is in C:\Program Files\FlightGear 3.4.0\bin\fgfs.exe

BojanKverh, 2015-06-12:
I understand, I will look into this issue

BojanKverh, 2015-06-12:
Ok, this is now fixed, but only if you not change the default location at FlightGear installation. I am not sure how to solve this universally, if the user doesn't have FlightGear in his path. Of course app could search through entire filesystem, but this is really time consuming operation.
The next issue here is that the arducopter is not present in the original FG installation, so FG will exit if called with --aircraft=arducopter parameter. If you could upload the files for this vehicle to GitHub or Dropbox, I could include it into the rtGCS installation package, so that it would be installed together with rtGSC.

dcmcshan, 2015-06-12:
Uploading to dropbox.  I had tried to commit to github, but I think it was too big!

dcmcshan, 2015-06-12:
If not found, we could simply open a dialog for user to specify the flight gear install directory

BojanKverh, 2015-06-12:
Yes, that a good idea

BojanKverh, 2015-06-12:
The installation package is prepared and uploaded to Dropbox. It includes arducopter files, which will be installed into default FlightGear folder.

BojanKverh, 2015-06-18:
With all the gauges moved into plugin library, I created a new installation package and uploaded it to Dropbox.

BojanKverh, 2015-06-20:
If application by itself doesn't find FG executable, it offers user a possibility to select (find) it via file dialog.