Syzygyx/DroneWidgets #28: On Windows, Fix Graphics Overlap

dcmcshan, 2015-06-12:
On Windows, the gauge graphics overlap the widget frame rather than scale to widget size.

dcmcshan, 2015-06-12:
Also, perhaps related, I can't seem to tile the gauges on Windows very effectively.  All in all this makes for a poor demonstration...

BojanKverh, 2015-06-12:
You mean the individual gauges, or the flight instruments widgets which shows all 6 of them? 
One of the problems is fixed minimal size. I set it up, because these gauges have to be of certain size, otherwise text and numbers will become unreadable. Secondly, if you are referring to the fact that if you put let say 4 gauges into the same dock area, they will likely not be of the same size. I am aware of that, but I am afraid the solution could only be that the gauges present somehow synchronize their size with each other - not an easy solution.

dcmcshan, 2015-06-12:
I’ll see if I can’t get a screenshot, but on your own system, try to reproduce this layout in Windows.  I could not.  

Can we size text and numbers with the window, perhaps to allow smaller minimum size?

BojanKverh, 2015-06-12:
Font size is actually getting smaller when you resize the gauge to a smaller size, but at some point the font will get too small to read.

BojanKverh, 2015-06-12:
Fonts are actually resized together with the gauge, but if you make a gauge really small, at some point the font will become unreadable.

BojanKverh, 2015-06-12:
OK, now I changed minimum gauge size from 150x150 to 75x75, which I think is the absolute minimum, where user can still read the text and to me it seems ok (checked on Windows), if there are still issues, please upload the screenshot.