Syzygyx/DroneWidgets #29: Android

dcmcshan, 2015-06-12:
For future reference, do you happen to have an Android phone or tablet?  At some point, we may want to port this to android, but also I have some other projects coming down the pipeline that might could use some quick Qt apps...

BojanKverh, 2015-06-12:
I have both, phone and tablet on Android. I did a few simple applications for it quite some time ago. I remember the Android Qt implementation used to be a little bit buggy, but hopefully things have changed with later releases.

BojanKverh, 2015-06-19:
Since I think I finished all the other issues, I will try to compile the application for Android. Please add any remaining issues.

dcmcshan, 2015-06-19:
Good.  Sorry, it's been a busy week.  I'll try and get some more tasks in the queue early next week

BojanKverh, 2015-06-19:
OK, no problem.

BojanKverh, 2015-06-20:
Managed to compile the app, however it crashes soon after startup (at least on simulator, didn't check on the real device). Will try to look more into it, probably just some lib doesn't get included or something like that.