Syzygyx/DroneWidgets #30: Plugin Widget Question

dcmcshan, 2015-06-15:
Possible to setup an architecture where we can compile the widgets separately without recompiling the whole executable?  So we'd have a base version of the QGC fork and just compile up widgets as we go along.

BojanKverh, 2015-06-15:
Yes, it is possible, although I have never done this before. Which widgets would you like to compile separately? Also, some code is already integrated in the original qgroundcontrol files, like MainWindow.cc and QGCMapWidget.cc, so that will be much more difficult to take out and compile as a separate plugin.

dcmcshan, 2015-06-15:
Just thinking ahead and wanting an easily extensible architecture for customers.  In general, we would have a main product and then be able to send them plugins for new features.  For instance, if we wanted to package this up for a different customer, would be great to have a "branding" package with the icons and splash as well as widgets specifically for their aircraft.  Then we wouldn't have to maintain an executable for each customer.

BojanKverh, 2015-06-15:
I understand, I think it is a good approach. I will keep that in mind.

BojanKverh, 2015-06-16:
With the extensive help from (very good) manuals found on Internet, I managed to add functionality to QGC through a plugin loader. Now loading the splash screen is done by ReferenceTechnologiesPlugin.so library. Now we could move gauges and maybe some other stuff also into plugins.

dcmcshan, 2015-06-16:
Cool.  I’ll figure out how we want to bundle things, but in general, maybe it makes sense to just put each widget in a plugin?

I’ve also moved syzygyx/qgroundcontrol to the syzygyx/zyg repositiory.  It will be private.  Let me know if you can use it.  

I think what I actually want to do is to have our qgroundcontrol repository just have the plugin loader changes, and all the widgets will live in the /zyg directory.  This way, we can keep qgroundcontrol updated as the other repo evolves.

BojanKverh, 2015-06-16:
I can access the zyg repo, except that the latest commit from today is missing (the one implementing plugins).

Regarding the plugin and qgroundcontrol repo separation; the idea is good and most of the widgets can be moved into the plugin repo. But I don't know about keeping qgroundcontrol updated: the problem is that some changes needed to be made in the original qgroundcontrol code in order to accommodate the changes we made and trying to keep the QGC code up to date will delete our changes, unless both branches cannot be merged somehow? Even if I try to move all my code out of QGC repo, at least the code that loads a plugin and generates missing functionality still has to stay there.

dcmcshan, 2015-06-16:
Right, there will have to be our plugin changes in the forked repo.  And we’ll have to merge from time to time, but that is what it is.

I can’t easily get your last commit over to the zyg repo, but sounds like what we want to do is to leave those changes in the qgroundcontrol directory anyways, and just put plugins in the zyg repo.

BojanKverh, 2015-06-16:
Yes, interfaces and plugins folder I think

BojanKverh, 2015-06-16:
OK, I transferred the plugins folder, interfaces I think it's better to leave it in qgroundcontrol

dcmcshan, 2015-06-16:
Okay, you can clean up the zyg repository.  It doesn't need all of qgroundcontrol in it

BojanKverh, 2015-06-17:
Today I will transfer as many widgets as possible into plugins. For other functionalities that still need to be done in the future, please feel free to open new issue threads.

BojanKverh, 2015-06-17:
All the flight indicators are now available as plugins. I think Mission planning, Vehicle widget and FlightGear video feed still remain to be ported.

BojanKverh, 2015-06-18:
Moved Mission planning, vehicle widget and FlightGear video feed widget into the plugin library. Please let me know, if there is anything else we should move there.

dcmcshan, 2015-06-22:
Should the zyg plugin library have the full qgrouncontrol app?  Does it need it to build?  If not, maybe we should clean it so that it just builds the plugins?

BojanKverh, 2015-06-22:
zyg plugin only needs interfaces folder from qgroundcontrol. I already thought about moving that into zyg. However, one has to depend on the other or the other way around, they cannot be two completely different entities.

BojanKverh, 2015-06-23:
Were you able to compile and test the app and plugins then?

dcmcshan, 2015-06-23:
Going to check it out on Windows today!

What do I build for the plugins?  Is there a Qt Creator file just for plugins?

BojanKverh, 2015-06-23:
It is in zyg/plugins/ReferenceTechnologies/ReferenceTechnologies.pro

BojanKverh, 2015-06-24:
Did you manage to build the app and plugins? There is also the installation package prepared for Windows on DropBox.

BojanKverh, 2015-07-04:
Hi Dan, how is it going?
I did a couple of corrections today on qgroundcontrol, that I haven't noticed so far. The plugin path on Linux should now be ok and also the plugin widgets are visible immediately after loading at startup.